walthowd/husbzb-firmware #4: Exec format error

pcwilcox, 2020-09-18:
Running the given command gives a format error from Docker:

```
$ docker run --rm --device=/dev/ttyUSB1:/dev/ttyUSB1 -it walthowd/husbzb-firmware
standard_init_linux.go:211: exec user process caused "exec format error"
```

I'm able to manually clone the repo, build it, and run it, so I'm not sure what exactly the problem is. This is on an RPi 4 running Ubuntu 20.04.

adrum, 2020-09-19:
Same error here on Raspberry Pi OS: 

`4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux`
`Docker version 19.03.11, build 42e35e6`

JMan7777, 2020-10-05:
Hi,

same error here on a Raspberry PI 4B 8GB.

```
root@raspberrypi:~# uname -a
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

root@raspberrypi:~# docker version
Client: Docker Engine - Community
 Version:           19.03.13
 API version:       1.40
 Go version:        go1.13.15
 Git commit:        4484c46
 Built:             Wed Sep 16 17:07:02 2020
 OS/Arch:           linux/arm
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.13
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.13.15
  Git commit:       4484c46
  Built:            Wed Sep 16 17:00:52 2020
  OS/Arch:          linux/arm
  Experimental:     false
 containerd:
  Version:          1.3.7
  GitCommit:        8fba4e9a7d01810a393d5d25a3621dc101981175
 runc:
  Version:          1.0.0-rc10
  GitCommit:        dc9208a3303feef5b3839f4323d9beb36df0a9dd
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683

root@raspberrypi:~# docker info
Client:
 Debug Mode: false

Server:
 Containers: 1
  Running: 0
  Paused: 0
  Stopped: 1
 Images: 2
 Server Version: 19.03.13
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 8fba4e9a7d01810a393d5d25a3621dc101981175
 runc version: dc9208a3303feef5b3839f4323d9beb36df0a9dd
 init version: fec3683
 Security Options:
  seccomp
   Profile: default
 Kernel Version: 5.4.51-v7l+
 Operating System: Raspbian GNU/Linux 10 (buster)
 OSType: linux
 Architecture: armv7l
 CPUs: 4
 Total Memory: 7.69GiB
 Name: raspberrypi
 ID: JHTA:RYUL:MP4M:JKKW:OK4U:GKC2:NC3D:EC4H:BSBK:CPOL:2ADZ:XI5C
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

root@raspberrypi:~# docker run --rm --device=/dev/ttyUSB1:/dev/ttyUSB1 -it walthowd/husbzb-firmware
standard_init_linux.go:211: exec user process caused "exec format error"
```

brikrav, 2020-10-15:
I was also getting this error, running SSH to a Raspberry Pi 4.  I was also able to manually clone the repo, build, and upgrade my firmware.

dadatuputi, 2020-10-17:
I overcame this on my Pi 4 by cloning the repo and building it locally: 

```bash
git clone https://github.com/walthowd/husbzb-firmware.git
cd husbzb-firmware
docker build -t walthowd/husbzb-firmware .
... wait 10 minutes or so for it to finish build ...
docker run --rm --device=/dev/ttyUSB1:/dev/ttyUSB1 -it walthowd/husbzb-firmware
...
```

CanDoAnything, 2020-10-28:
@dadatuputi , any thoughts? do i need to clone a specific branch?

``` console
pi@raspberrypi4:~/husbzb/husbzb-firmware $ docker build -t walthowd/husbzb-firmware .
Sending build context to Docker daemon  3.749MB
Step 1/12 : FROM ubuntu:18.04
 ---> 32114f98a2fc
Step 2/12 : COPY qemu-arm-static /usr/bin
COPY failed: stat /var/lib/docker/tmp/docker-builder128520292/qemu-arm-static: no such file or directory
```

dadatuputi, 2020-10-28:
> @dadatuputi , any thoughts? do i need to clone a specific branch?
> 
> ```
> pi@raspberrypi4:~/husbzb/husbzb-firmware $ docker build -t walthowd/husbzb-firmware .
> Sending build context to Docker daemon  3.749MB
> Step 1/12 : FROM ubuntu:18.04
>  ---> 32114f98a2fc
> Step 2/12 : COPY qemu-arm-static /usr/bin
> COPY failed: stat /var/lib/docker/tmp/docker-builder128520292/qemu-arm-static: no such file or directory
> ```

Looks like the repo owner made a commit that added a few lines: https://github.com/walthowd/husbzb-firmware/commit/f09251f135745984274a789b541dbba4a475b259

Look at the last line of hooks/post_checkout, it downloads qemu-arm-static. But @walthowd should know that building the docker image manually is broken and either fix it (include the download of qemu in the dockerfile, for example) or post instructions on how to build it.

flockofseagolls, 2020-10-28:
> > @dadatuputi , any thoughts? do i need to clone a specific branch?
> > ```
> > pi@raspberrypi4:~/husbzb/husbzb-firmware $ docker build -t walthowd/husbzb-firmware .
> > Sending build context to Docker daemon  3.749MB
> > Step 1/12 : FROM ubuntu:18.04
> >  ---> 32114f98a2fc
> > Step 2/12 : COPY qemu-arm-static /usr/bin
> > COPY failed: stat /var/lib/docker/tmp/docker-builder128520292/qemu-arm-static: no such file or directory
> > ```
> 
> Looks like the repo owner made a commit that added a few lines: [f09251f](https://github.com/walthowd/husbzb-firmware/commit/f09251f135745984274a789b541dbba4a475b259)
> 
> Look at the last line of hooks/post_checkout, it downloads qemu-arm-static. But @walthowd should know that building the docker image manually is broken and either fix it (include the download of qemu in the dockerfile, for example) or post instructions on how to build it.

I want to preface my response by saying that I know nothing about any of this stuff, but I got around this issue by running /husbzb-firmware/hooks/post_checkout. Then I copied the qemu-arm-static file that was generated in /husbzb-firmware/hooks to /husbzb-firmware and reran "docker build -t walthowd/husbzb-firmware ."

CanDoAnything, 2020-10-31:
@flockofseagolls , that worked for me. 

## Summary of steps

1. 🛑 Stop Home Assistant's container to make sure nothing else would be accessing the USB stick during fw update
1. 👨‍💻  `git clone https://github.com/walthowd/husbzb-firmware.git`
1. 👨‍💻 `cd husbzb-firmware`
1. 👨‍💻 `curl -L https://github.com/balena-io/qemu/releases/download/v3.0.0%2Bresin/qemu-3.0.0+resin-arm.tar.gz | tar zxvf - -C . && mv qemu-3.0.0+resin-arm/qemu-arm-static .`
1. 👨‍💻 `docker build -t walthowd/husbzb-firmware .` 
    - (This process took a while...at least 30++ minutes on my raspi4 running raspbian. @dadatuputi were you already on debian? The script unpacked and installed ~200 packages for me)
1. 🛑 Make sure Home assistant is stopped
1. 👨‍💻`docker run --rm --device=/dev/ttyUSB1:/dev/ttyUSB1 -it walthowd/husbzb-firmware bash`
1. 👨‍💻 `python ncp.py flash -p /dev/ttyUSB1 -f ncp-uart-sw-6.6.3.ebl `
1. 👨‍💻 `python ncp.py scan` --> returns --> `{"ports": [{"stackVersion": "6.6.3-151", "deviceType": "zigbee", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}]}`
1. ❓  I ran this upgrade in hopes it would solve my issue here <https://github.com/zigpy/bellows/issues/179#issuecomment-719169072>
1. 😭 Unfortunately, it didn't 
[image]

RandPooka, 2020-11-22:
How do you use this without docker, though?

I see the ebl files, and I can find the appropriate port settings in ncp.py

But, all the scripts and single commands are docker specific. I want to update the firmware using openhabian on my pi in a non docker form

Vartkat, 2021-01-04:
> I overcame this on my Pi 4 by cloning the repo and building it locally:
> 
> ```shell
> git clone https://github.com/walthowd/husbzb-firmware.git
> cd husbzb-firmware
> docker build -t walthowd/husbzb-firmware .
> ... wait 10 minutes or so for it to finish build ...
> docker run --rm --device=/dev/ttyUSB1:/dev/ttyUSB1 -it walthowd/husbzb-firmware
> ...
> ```

This one worked for me on Rasp Pi 4.
Thanks

Kezzsim, 2021-02-14:
Anyone doing this on a Pi4 with the latest Raspbian might need to custom upgrade their version of libsecomp before continuing. If you get an error about failed gpg signatures when trying to build the docker image, run
`wget http://http.us.debian.org/debian/pool/main/libs/libseccomp/libseccomp2_2.5.1-1_armhf.deb`
(_apt-get install wget if you need it, or use cURL_)
followed by
`sudo dpkg -i libseccomp2_2.5.1-1_armhf.deb`
in that same directory, then try `docker build -t walthowd/husbzb-firmware .` again

devjj, 2021-02-23:
The author just merged my pull request that includes a Dockerfile for RPi4. I used it successfully to upgrade my stick.

kan84, 2021-02-27:
Any way to do this on pi 3b+? Tried manually building locally but that failed as well

sreknob, 2021-04-07:
@kan84 try outside the docker image but you will need to install dependencies. Depending on the OS version you're running you might have to specify python 2.7 and use sudo.
```
git clone https://github.com/walthowd/husbzb-firmware.git
cd husbzb-firmware
pip install xmodem
pip install pyserial
./ncp.py flash -p /dev/ttyUSB1 -f ncp-uart-sw-6.7.8.ebl
```