walthowd/husbzb-firmware #5: Mistake in the manual?

airens, 2020-10-01:
Hello, tried to update my stick in manual mode. You say, that I need to run the command `./ncp.py flash -p /dev/ttyUSB1 -f ncp-uart-sw-6.6.3.ebl`, but there is no such a file with firmware. In my Docker container I only have `ncp-uart-sw.ebl`

cboylan, 2020-10-10:
It looks like the docker hub hosted image hasn't been updated since https://github.com/walthowd/husbzb-firmware/commit/0ab67453a1e552706c260a4c83a6ab7bfe988917 was committed to the repo. That commit renamed `ncp-uart-sw.ebl` to `ncp-uart-sw-6.6.3.ebl`.

You can confirm this checking the sha1sums of the `ncp-uart-sw.ebl` file in the docker image and at https://github.com/walthowd/husbzb-firmware/raw/master/ncp-uart-sw-6.6.3.ebl

If @walthowd rebuilds and updates the image on docker hub I think this problem goes away. In the meantime you can use `ncp-uart-sw.ebl` in the existing image or build the docker image yourself locally and use that which should have all the .ebl files present in the root of the git repo.

Vlad-Star, 2020-10-30:
> Note! Please understand that as of September 2020, the 6.x.x releases and higher of SiLabs EmberZNet will require Home Assistant 0.115 or higher.

@walthowd sorry for asking - could you please clarify:
1) Would HA versions 0.114 (and below) work with 6.x.x stick firmware (i.e. backward compatible) or only work with 5.x.x?
2) Is there a way to update firmware on Hassio/RPi4 with connected Nortek stick "in place" (i.e. without a docker)?