walthowd/husbzb-firmware #6: Can't connect to my device

airens, 2020-10-01:
Hello, trying to update my ZigBee adapter with your tool and something going wrong, when  ncp.py trying to connect to my stick. When I run `./ncp.py scan` it says "Connecting to.. /dev/ttyUSB1 57600 True False" and then nothing happens until I hit Ctrl-C and get the next:
`
^CException: <type 'exceptions.KeyboardInterrupt'> /dev/ttyUSB1
{"ports": [{"deviceType": "unknown", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}]}
`
So, it looks like the tool sees my adapter, but freezes on connecting to it. Any thoughts?

pruchai, 2021-04-18:
I have the same issue in Ubuntu running on a Nuc. 

Nothing else is using the port. The stick is HUSBSZ-1. Works fine under Home Assistant

ur7x, 2021-06-13:
Mine does this too.  I ignored this a did the update step.. and it worked. Well I think it worked since it still does not report the version.

MattWestb, 2021-06-13:
If running in HA is Bellows writing the version in the long at startup so you can see it there or using bellows and command info.