walthowd/husbzb-firmware #10: Unable to launch shell

d4v3d, 2020-11-30:
Hi,

I am trying to use this firmware updater but running into a problem on my setup. I am running the software in Docker on my Unraid server.

I have successfully installed the software and verified that it can communicate with the USB stick because when I check the logs I see the following message:
`{"ports": [{"stackVersion": "5.4.1-194", "deviceType": "zigbee", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}]}
Found zigbee port at /dev/ttyUSB1 running 5.4.1-194`

However, when I try to launch the console/shell to perform the firmware update I get the following error written to my Unraid system log exactly eight times:
`Nov 29 21:52:06 Tower nginx: 2020/11/29 21:52:06 [error] 8802#8802: *28409122 connect() to unix:/var/tmp/husbzb-firmware.sock failed (111: Connection refused) while connecting to upstream, client: 192.168.1.213, server: , request: "GET /dockerterminal/husbzb-firmware/ws HTTP/1.1", upstream: "http://unix:/var/tmp/husbzb-firmware.sock:/ws", host: "tower"`

Any ideas on what I should try to resolve this error?

Thanks!
Dave

sreknob, 2020-12-01:
Hi Dave -

This is an unRAID error connecting to the container rather than an issue with the container. Could be an emhttp issue, what version are you running? Anyway, best to go the to unRAID forums for additional help if this is happening with all your containers.

As a workaround, it's best to just enter the container directly from an SSH session on your unRAID box as below, then run the commands in the readme.

1. start ssh session to unRAID (either webui or other client)
2. make sure your container is running in your unRAID webui
3. run the command `docker exec -it husbzb-firmware bash`
4. follow the readme

>  ```
>  cd /tmp/silabs
>  ./ncp.py flash -p /dev/ttyUSB1 -f ncp-uart-sw-6.6.5.ebl
>  ```

Good luck!

d4v3d, 2020-12-01:
Ah, thank you! You put me on the right path! I was trying to launch from the Docker UI page but I ended up just running from the Unraid console.

[image]

All updated now!

closed 2020-12-01 by d4v3d