walthowd/husbzb-firmware #11: Unable to run on Mac

rodhall, 2020-12-28:
I have not been able to get this to work on a Mac where I run HA and have the HubZ installed. The issue is finding the correct /dev/XXXX path I think but not sure?

jjmuriel, 2021-02-10:
I unpluged the dongle and installed the driver (macOS Bi Sur) from http://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers.  Now, I have 2 new tty:  **tty.GoControl_zigbee**  and **tty.GoControl_zwave** , but when I execute the next command:  
`docker run --rm --device=/dev/tty.GoControl_zigbee:/dev/tty.GoControl_zigbee -it walthowd/husbzb-firmware`

I get this error:  
`docker: Error response from daemon: error gathering device information while adding custom device "/dev/tty.GoControl_zigbee": no such file or directory`

walthowd, 2021-02-10:
I don't believe Mac for docker supports device mapping for serial ports. You may have some luck if you run ncp.py directly on hardware (you need Python 2.7ish, which I believe is still avaialble on Mac).

jjmuriel, 2021-02-10:
Thank you @walthowd , but when I execute `% python2.7 ncp.py scan` with the HUSBSZ-1 dongle plugged,  I get this error:  
`Traceback (most recent call last):
  File "ncp.py", line 12, in <module>
    from xmodem import XMODEM
ImportError: No module named xmodem`

Using sudo I get the same.

walthowd, 2021-02-10:
Yes, you'll need to install `xmodem` and `pyserial` modules for Python 2 as a prerequisite.

jjmuriel, 2021-02-10:
After install xmodem and pyserial (pip2.7) I got this:

`Traceback (most recent call last):
  File "ncp.py", line 16, in <module>
    import serial.tools.list_ports
  File "/usr/local/lib/python2.7/site-packages/pyserial-3.4-py2.7.egg/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/usr/local/lib/python2.7/site-packages/pyserial-3.4-py2.7.egg/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/usr/local/lib/python2.7/site-packages/pyserial-3.4-py2.7.egg/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found`

walthowd, 2021-02-10:
Bummer, looks like a pyserial bug:

https://github.com/pyserial/pyserial/issues/509

Might just be easiest to go the Virtual Box route and run Ubuntu iso temporarily.

jjmuriel, 2021-02-10:
I installed "comports" also, but i get the same.  Yes, I'll try with VirtualBox/Linux. Thank you @walthowd.

jjmuriel, 2021-02-10:
With Kali on Virtual Box everything was OK:
After:
`Connecting to.. /dev/ttyUSB1 57600 True False 
{"ports": [{"stackVersion": "5.4.1-194", "deviceType": "zigbee", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}]}`

Before:
`Connecting to.. /dev/ttyUSB1 57600 True False 
{"ports": [{"stackVersion": "6.6.5-204", "deviceType": "zigbee", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}]}`

Thank you @walthowd , nice job!

aicarmic, 2023-05-09:
For anyone reading this and trying to run on MacOS, you have to first install the USB to UART bridge drivers from Sililabs: https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads

Once you've install this, it should map both the zwave and zigbee radios to a dev/tty port. Mine looked like this:

```
/dev/tty.GoControl_zigbee?
/dev/tty.GoControl_zwave
```

Hope this helps someone!

TrumpetH4X, 2023-05-10:
If it helps anyone else, my zigbee delivice was listed as something really weird like "/dev/cu.GoControl_zigbee\n" when I ran ./ncp.py. Had to use ls -l /dev to find out the *actual* device name was /dev/cu.GoControl_zigbee$'\r'.

Pasted that in and everything worked pronto!