walthowd/husbzb-firmware #15: Upgrade hangs at "CEL stick" step

eric72777, 2021-01-23:
I was unable to upgrade a HUSBZB-1 (5.4.1-192) using the docker image. Restarting the computer and unplugging and re-plugging the stick didn't fix the issue.

>sudo docker run --rm --device=/dev/ttyUSB1:/dev/ttyUSB1 -it walthowd/husbzb-firmware bash
>
>cd /tmp/silabs
>
>./ncp.py scan
>Connecting to.. /dev/ttyUSB1 57600 True False 
>{"ports": [{"stackVersion": "5.4.1-194", "deviceType": "zigbee", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}]}
>./ncp.py flash -p /dev/ttyUSB1 -f ncp-uart-sw-6.6.5.ebl 
>Restarting NCP into Bootloader mode...
>CEL stick
>^CTraceback (most recent call last):
>  File "./ncp.py", line 459, in <module>
>    flash(args.port,args.file)
>  File "./ncp.py", line 310, in flash
>    ser.readline() # read blank line
>  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 565, in read
>    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
>KeyboardInterrupt

ajmath, 2021-02-20:
I'm seeing the same issue.

MikePaer, 2021-04-07:
Same problem here. Also, running the docker run main command just hangs. Has anyone figured this out?

sreknob, 2021-04-07:
The next step in the script is the initialization of the serial port which might be failing... for me re-plugging fixed it.
If you can open up your stick, try the firmware recovery instructions in the readme as a work around. It's actually quite easy to do it that way.
The other thing you could try is another linux environment. If you clone the repo and install python 2.7 and xmodem, you can use the python script outside of a container. I've done this on just a raspberry-pi rather than running the docker image.

MikePaer, 2021-04-08:
Thanks for the pointers. Tried running the Python script outside the container and on a different machine, neither worked. I didn’t try the firmware recovery, and instead replaced my HUSBZB-1 device assuming it was defective. The new device worked right out of the box with no issues. I think I just had a lemon.

C2BB, 2022-08-16:
Thanks @sreknob your tip on unplugging the stick solved the issue for me.

BHSPitMonkey, 2023-04-09:
Having the same issue today with one of my HUSBZB-1 devices. Note that I am commenting out `xonxoff=True` as a workaround for the other issues reported in different threads (this has helped me flash another device previously), but in this case the failure occurs either way.

Here's some `strace` output indicating what's going on behind the pyserial `readline()` call that never finishes (never sees a `\n` appear):

https://github.com/walthowd/husbzb-firmware/blob/12de9886507b4f806e655c839a36e64f237fe8fe/ncp.py#L296-L299

```
write(3, "\32\3008\274~", 5)            = 5
read(3, "\32\301\2\v\nR~", 7)           = 7
write(3, "\0B!\250\\,\240~", 8)         = 8
read(3, "\1B\241\250P(\5\346\177b~", 11) = 11
write(3, "\201`Y~", 4)                  = 4
write(3, "\32\3008\274~", 5)            = 5
read(3, "\32\301\2\v\nR~", 7)           = 7
write(3, "\0B!\250S\335O~", 8)          = 8
read(3, "\1B\241\250P(\5\346\177b~", 11) = 11
write(3, "\201`Y~", 4)                  = 4
write(3, "\32\3008\274~", 5)            = 5
read(3, "\32\301\2\v\nR~", 7)           = 7
write(3, "\0B!\250R\315n~", 8)          = 8
read(3, "\1B\241\250P(\5\346\177b~", 11) = 11
write(3, "\201`Y~", 4)                  = 4
write(3, "\32\3008\274~", 5)            = 5
read(3, "\32\301\2\v\nR~", 7)           = 7
write(3, "\0B!\250Q\375\r~", 8)         = 8
read(3, "\1B\241\250P(\5\346\177b~", 11) = 11
write(3, "\201`Y~", 4)                  = 4
write(3, "\32\3008\274~", 5)            = 5
read(3, "\32\301\2\v\nR~", 7)           = 7
write(3, "\0B!\250P\355,~", 8)          = 8
read(3, "\1B\241\250P(\5\346\177b~", 11) = 11
write(3, "\201`Y~", 4)                  = 4
```

https://github.com/walthowd/husbzb-firmware/blob/12de9886507b4f806e655c839a36e64f237fe8fe/ncp.py#L296-L299

```
write(3, "}1C!'Un\220~", 9)             = 9
```

https://github.com/walthowd/husbzb-firmware/blob/12de9886507b4f806e655c839a36e64f237fe8fe/ncp.py#L309-L311

```
read(3, "\22C\241'T\2539~", 8)          = 8
```

https://github.com/walthowd/husbzb-firmware/blob/12de9886507b4f806e655c839a36e64f237fe8fe/ncp.py#L316-L326

```
write(3, "\n", 1)                       = 1
```

https://github.com/walthowd/husbzb-firmware/blob/12de9886507b4f806e655c839a36e64f237fe8fe/ncp.py#L327-L330

```
read(3, "\230", 1)                      = 1
read(3, "`", 1)                         = 1
read(3, "\0", 1)                        = 1
read(3, "\30", 1)                       = 1
read(3, "\30", 1)                       = 1
read(3, "\236", 1)                      = 1
read(3, "x", 1)                         = 1
read(3, "\376", 1)                      = 1
read(3, "\206", 1)                      = 1
read(3, "\376", 1)                      = 1
read(3, "\6", 1)                        = 1
read(3, "\340", 1)                      = 1
read(3, "\6", 1)                        = 1
read(3, "\340", 1)                      = 1
```

(At this point the script hangs indefinitely and there's no more reads/writes. Pyserial waits indefinitely for a newline character but only receives the unexpected data shown in the `strace` above.)