walthowd/husbzb-firmware #20: Does this also update Zwave firmware

tehbra1n, 2021-03-16:
More of a question than an issue.  First thanks for the great work here, I just updated my HUSBZB stick w/o issue.   Doing the reading on it and digging around I'm getting the feeling that this might only upgrade the ZigBee firmware on the stick, leaving the ZWave firmware untouched.

Am I totally misinterpreting what I'm reading?

sreknob, 2021-03-16:
You are correct, this only updates the Zigbee firmware and does not modify the z-wave firmware. There is no update for the z-wave side that I am aware of.

walthowd, 2021-03-16:
You can follow this guide to update the z-wave side. It does require Windows and some precise shorting of test points:  

https://community.home-assistant.io/t/husbzb-1-z-wave-firmware-portion-update/259623

tehbra1n, 2021-03-16:
> You can follow this guide to update the z-wave side. It does require Windows and some precise shorting of test points:
> 
> https://community.home-assistant.io/t/husbzb-1-z-wave-firmware-portion-update/259623

Perfect, thanks.

closed 2021-03-16 by tehbra1n