walthowd/husbzb-firmware #22: Firmware upgrades and downgrades for the new ITead Zigbee 3.0 USB Dongle

Hedda, 2021-03-30:
Suggest add newer + older firmware for the new EFR32MG21 based Zigbee 3.0 USB Dongle Model 9888010100045 from ITead

https://www.itead.cc/zigbee-3-0-usb-dongle.html

https://github.com/xsp1989/zigbeeFirmware

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle

https://github.com/xsp1989/zigbeeFirmware/blob/master/README.md

Hedda, 2021-03-30:
The same firmware developer also hosts firmware the "SM-011 V1.0" modules by CoolKit-Technologies on his GitHub repo:

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/SM-011

"SM-011 V1.0" modules can be found on Aliextress and is used in some DIY Zigbee dongles and gateways:

https://www.aliexpress.com/item/4001019552844.html

https://modkam.ru/?p=1889 (description in Russian, use a translator)

Looks like @CoolKit-Technologies = Shenzhen Cool House Technology Co., Ltd. (abbreviated as Cool House Technology) might be the Chinese manufacturer of this SM-011 V1.0 / ZYZBP008 module:

https://www.coolkit.cn/product/sm-011/

https://github.com/CoolKit-Technologies/DevDocs/tree/master/Zigbee

https://github.com/CoolKit-Technologies/DevDocs/blob/master/Zigbee/SM-011%E5%BA%94%E7%94%A8%E6%8C%87%E5%AF%BC%E4%B9%A6.md

https://translate.google.com/translate?hl=&sl=auto&tl=en&u=https%3A%2F%2Fgithub.com%2FCoolKit-Technologies%2FDevDocs%2Fblob%2Fmaster%2FZigbee%2FSM-011%E5%BA%94%E7%94%A8%E6%8C%87%E5%AF%BC%E4%B9%A6.md

"SM-011 V1.0" / ZYZBP008 model is a EFR32 MG21 (i.e. Silicon Labs EFR32 Series 2 ) Zigbee radio module with GPIO for DIY:

[image] 
[image]

That ZYZBP008 / SM-011 V1.0 module is sold here on Aliexpress for about $12 (US) including free shipping:

https://www.aliexpress.com/item/4000592757948.html

https://www.aliexpress.com/item/1005002081321746.html

ZYZBP008 Zigbee Gateway (SM-011 V1.0 module-based) with EFR32MG21 Zigbee radio:

[image]

Matching USB dongle/adapter based on ZYZBP008/SM-011 V1.0 is sold on Aliexpress for about $18 (US) including free shipping:

https://www.aliexpress.com/item/4001019552844.html

https://www.aliexpress.com/item/1005002023447724.html

ZB-GW02 (SM-011 V1.0 module-based) gateway looks to be Tuya type WiFi gateway with "SM-011 V1.0" module pre-soldered.

[image]

That ZB-GW02 Zigbee gateway is sold here on Aliexpress for $27.58 (US) including free shipping:

https://www.aliexpress.com/item/4001019552844.html

The USB dongle/adapter, WiFi-gateway, all use this "SM-011 V1.0" module which can also be purchased separately. It is based on Silicon Labs EFR32MG21A020F768IM32 radio chip that has 'only' have 64KB RAM and 768KB FLASH storage, which is slightly less RAM and FLASH storage than something like the official SLWRB4180A which is based on EFR32MG21A020F1024IM32 (that has 96KB RAM and 1024KB FLASH storage), however, it still features powerful a Tx power 20dBm radio transceiver as well as an uFL/u.FL/IPX/IPEX/UMCC Female connector for connecting an optional external antenna.

MattWestb, 2021-03-30:
If @walthowd like having it here its OK and then putting on link to it from Zigpy Wiki for the device. 
If hi only like having it for EM35X devices then do one larger section on Zigpx Wiki as is don for every known documented radio coordinator device.

And first putt it in the Zigpy coordinator list in the Wiki.

Hedda, 2021-04-01:
Would also be great it could get a Zigbee Router firmware for ITead Zigbee 3.0 USB Dongle and "SM-011 V1.0" as well.

https://github.com/grobasoz/zigbee-firmware/issues/11

https://github.com/xsp1989/zigbeeFirmware/issues

walthowd, 2021-04-01:
Seems to work for flashing

```
root@f546c76e9ab3:/tmp/silabs# ./ncp.py scan

Connecting to.. /dev/ttyUSB2 115200 True False 
{"ports": [{"stackVersion": "6.7.8-373", "deviceType": "zigbee", "pid": "7523", "port": "/dev/ttyUSB2", "vid": "1A86"}]}

root@f546c76e9ab3:/tmp/silabs# ./ncp.py flash -p /dev/ttyUSB2 -f /data/ncp-sonoff-uart-nsw_679_115200.gbl

Restarting NCP into Bootloader mode...
CEL stick
Gecko Bootloader v1.9.2

Successfully restarted into bootloader mode! Starting upload of NCP image... 
Finished!
Rebooting NCP...
root@f546c76e9ab3:/tmp/silabs# ./ncp.py scan
Connecting to.. /dev/ttyUSB2 115200 True False 
{"ports": [{"stackVersion": "6.7.9-405", "deviceType": "zigbee", "pid": "7523", "port": "/dev/ttyUSB2", "vid": "1A86"}]}
```

walthowd, 2021-04-01:
Seeing the same issue reported elsewhere, not able to join any devices-- tried with stock 6.7.8 and 6.7.9 from https://github.com/xsp1989/zigbeeFirmware