walthowd/husbzb-firmware #27: Scan Not reporting Version

ur7x, 2021-06-13:
The update ultimately worked (I think)
But when I ran the scan tool (./ncp.py scan)
It hung at USB0 (needed a cntl c)
And then it  reported this:

{"ports": [{, "deviceType": "unknown", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}.....[snip]

The update reported that it was successful, and the HUSBZB is working perfectly back on the HA Pi4,  but even after the firmware update it refused to report its version. 

To do the version update I used a second Rasperry Pi4.  Running a "full version" of Raspberrian and Python3

timgilbert, 2021-06-15:
I'm seeing the same issue on a little Celeron-powered micro-PC (running Manjaro). If I only pass in `USB1` it hangs for a while and then I see this:

```
% docker run --rm --device=/dev/ttyUSB1:/dev/ttyUSB1 -it walthowd/husbzb-firmware bash
root@9bd765282a2f:/tmp/silabs# ./ncp.py scan
Connecting to.. /dev/ttyUSB1 57600 True False
No ZigBee Ack. /dev/ttyUSB1
{"ports": [{"port": "/dev/ttyUSB1", "vid": "10C4", "pid": "8A2A", "deviceType": "unknown"}]}
```

MattWestb, 2021-06-15:
I can not saying anything about NCP.PY but if you is using bellows CLI or starting HA / ZHA is writing the version in the long like:
```
2020-08-30 10:17:39 INFO (MainThread) [bellows.zigbee.application] EZSP Radio manufacturer: IKEA of Sweden
2020-08-30 10:17:39 INFO (MainThread) [bellows.zigbee.application] EZSP Radio board name: Billy EZSP by MW
2020-08-30 10:17:39 INFO (MainThread) [bellows.zigbee.application] EmberZNet version: 6.7.6.0 build 327
```
From HA log but bellows is showing the same and little bit more.

timgilbert, 2021-06-15:
Here's what I get when running the python script directly on the OS instead of through docker:

```
(venv) % ./ncp.py scan
Connecting to.. /dev/ttyUSB1 57600 True False
Exception: <class 'serial.serialutil.SerialException'> /dev/ttyUSB1
Connecting to.. /dev/ttyUSB0 57600 True False
Exception: <class 'serial.serialutil.SerialException'> /dev/ttyUSB0
{"ports": [{"port": "/dev/ttyUSB1", "vid": "10C4", "pid": "8A2A", "deviceType": "unknown"}, {"port": "/dev/ttyUSB0", "vid": "10C4", "pid": "8A2A", "deviceType": "unknown"}]}
```

I haven't tried flashing the firmware yet. A month or two ago I was able to flash successfully, but I'm trying to move HA to a different platform, and on the new machine I'm seeing errors in the ZHA logs when running HA: `[bellows.ezsp] NCP entered failed state. Requesting APP controller restart`. Those errors are what prompted me to see if a newer firmware was available that might fix the problem.

MattWestb, 2021-06-15:
The comport speed is around line 32-37
```
# Nortek/GoControl HUSBZB-1
CEL_VID = '10C4'CEL_PID = '8A2A'
CEL_BAUD = 57600
CEL_XON_XOFF = True
CEL_RTS_CTS = False
```
Trying putting `115200` in the` CEL_BAUD = `

timgilbert, 2021-06-27:
Not really sure what's up with this, but when I set the baud to 115200 the `ncp.py scan` bit hangs when it's testing ttyUSB1 (which is the z-wave port), and when I set back to 57600 it eventually reports "No ZigBee Ack" from ttyUSB1 (expected, since it's z-wave) and then hangs when connecting to ttyUSB0. 

I'm no longer seeing the SerialException, but I think it was caused because my user was not originally in the `uucp` group (in Arch, this is the group with permission to open serial devices, apparently). I'm now testing in single-user mode so permissions shouldn't be a factor. ZHA is still giving me the same error about "NCP entered failed state" though.

timgilbert, 2021-06-27:
Ah, digging more deeply into my logs I'm seeing a kernel error message `cp210x 1-2:1.1: Failed to attach ACPI GPIO chip`, so possibly my problems here are BIOS/kernel related. Will do some spelunking on that.

dadatuputi, 2021-11-18:
Since ncp.py was updated to support Python 2 and 3, it doesn't work right with the current version of Python 2.7 on Raspbian:

```bash
hass@homeassistant:~/husbzb-firmware $ sudo ./ncp.py scan
Connecting to.. /dev/ttyUSB1 57600 True False
Exception: <type 'exceptions.TypeError'> /dev/ttyUSB1
Connecting to.. /dev/ttyUSB0 57600 True False
^CException: <type 'exceptions.KeyboardInterrupt'> /dev/ttyUSB0
Exception: <type 'exceptions.IndexError'> /dev/ttyAMA0
{"ports": [{"deviceType": "unknown", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}, {"deviceType": "unknown", "pid": "8A2A", "port": "/dev/ttyUSB0", "vid": "10C4"}, {"deviceType": "unknown", "port": "/dev/ttyAMA0"}]}
```

If I revert to the commit before ncp.py was updated, or if I use Python 3 it works properly.