walthowd/husbzb-firmware #28: Notes: Regarding Scan(Nortek HUSBZB-1) : unkown version, loong hangs while running and fails

davi1td, 2021-07-21:
Hi, thanks for such an awesome thing ...!
I had a hard time running scan, here are some of my notes, hopefully helps, if not ok ...
In both windows and running debian buster in a VmWare VM I had the same problem, scan would run forever and eventually fail.
In both cases I found running python 2.7 does NOT work.
You must be running python 3.X 
must use module pyserial not serial!, remove serial then add pyserial  : 
pip3 uninstall serial
pip3 install pyserial xmodem
i modified the script as follows:

def scan():
    outjson = {'ports':[]}
    for port in serial.tools.list_ports.comports():
     **if port.location is None or port.location!='2-2.2:1.1' : continue #===>optional what I found between windows and linux but could be different**

This was most import to speed things up :: 
                # Init serial port
                ser = serial.Serial(
                    **timeout=2,
                    write_timeout=2,**

then **python3 ncp.py scan** hopefully works :)
Regardless of running for Nortek HUSBZB-1, i believe the timeouts are essential.
Hope this helps!, thanks again for providing this~!
Regards,
-TD

MattWestb, 2021-07-21:
Thanks for findings and fixes !!

The `ncp-uart-sw-6.7.8-115k.ebl` is with software flow control then its working OK on EFR32MG and also on EM35X devices and many dont have hardware pins for HWFC and most user like having the double speed on the communication line.

davi1td, 2021-07-21:
Thanks for the confirmation.

closed 2021-07-21 by davi1td

MattWestb, 2021-07-21:
If you like you can using the `ncp-uart-sw-6.7.8.ebl` that is more standard with SWFC and 57.6 kb and is auto detected by ZHA (11.52 Kb is not for the moment) but its not make much sens only loosing the speed.

And ZHA have starting supporting HW FC but its not good tested and is not making so much sens then we have not cooking one EZSP 6.7.8.0 with hardware flow control and its not needed in over 95% of user cases.

davi1td, 2021-07-21:
Gotchya, ya already upgraded to ncp-uart-sw-6.7.8-115k.ebl with software flow control, working fine in openHab 3.2.
Thanks again.