walthowd/husbzb-firmware #29: hangs after printing bootloader version, never updates

billyburly, 2021-07-23:
I'm trying to update my HUSBZB-1. Used `sudo ./ncp.py flash -p /dev/ttyUSB1 -f ncp-uart-sw-6.7.8-115k.ebl` and had it hang after the reading the bootloader verison.

```
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962
```
If I interrupt it, I get the following traceback
```
Traceback (most recent call last):
  File "/home/durrw/husbzb-firmware/./ncp.py", line 487, in <module>
    flash(args.port,args.file)
  File "/home/durrw/husbzb-firmware/./ncp.py", line 353, in flash
    ser.readline() # BL > 1
  File "/usr/lib/python3.9/site-packages/serial/serialposix.py", line 483, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
KeyboardInterrupt
```

billyburly, 2021-07-23:
I was able to break out into the bootloader by adding an `exit()` after `print(verBL.decode('ascii')) # show Bootloader version` and upload the file via minicom. looks like something is broken waiting for the upload start prompt

lefebp, 2022-03-31:
I have the same issue, I guess I'll try to see how to use minicom.

lefebp, 2022-03-31:
> I have the same issue, I guess I'll try to see how to use minicom.

For others with this issue, this solved it for me: 
https://github.com/walthowd/husbzb-firmware/issues/33