walthowd/husbzb-firmware #30: Inconsistency in readme.md

Hydro-guy, 2021-08-06:
Readme.md states:
For advanced users - An alternative 115200 bps firmware image is available for HUSBZB-1, ncp-uart-sw-6.7.8-115k.ebl. This provides a faster pathway to the adapter but requires that new Home Assistant user manually input the serial path, radio type (EZSP) and bauddate when adding ZHA. Existing Home Assistant users will have to backup and edit the .storage/.core.config_entries file and update the listed baud rate

My .storage/.core.config_entries file does not contain any reference to HUSBZB-1 baud rate.  Raspberry pi 3 home assistant os 6.2 ver core-2021.8.1.  See below:
"entry_id": "d210882a593c42eeb9d3e5ddc87366cf",
                "version": 1,
                "domain": "zwave",
                "title": "Z-Wave",
                "data": {
                    "network_key": "",
                    "usb_path": "/dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_C0F003F0-if00-port0"
                },
                "options": {},
                "pref_disable_new_entities": false,
                "pref_disable_polling": false,
                "source": "user",
                "unique_id": null,
                "disabled_by": null
            },
            {
                "entry_id": "e8ba308ef3ab48a8ab627b59ec63f35c",
                "version": 2,
                "domain": "zha",
                "title": "/dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_C0F003F0-if01-port0",
                "data": {
                    "device": {
                        "path": "/dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_C0F003F0-if01-port0"
                    },
                    "radio_type": "ezsp"
                },
                "options": {
                    "custom_configuration": {
                        "zha_options": {
                            "enable_identify_on_join": true
                        }
                    }
                },

Guessing that the correct syntax will be:"baud_rate": 115200, entered in both zwave "data" field and zha "data" fields.  Haven't confirmed that yet.

Wouldnt have known syntax except that one of the integrations (mysensors) uses it with this format.

GeoffMaciolek, 2021-11-01:
Some help for anyone trying to read this: here's the isse creato's JSON as it should be (with guessed header)
```json
{
  "THIS_CONTEXT": "IS_JUST_A_GUESS",
  "version": 1,
  "key": "core.device_registry",
  "data": {
    "devices": [
      {
        "THE_ORIGINAL_ENTRY": "IS BELOW HERE"
      },
      {
        "entry_id": "d210882a593c42eeb9d3e5ddc87366cf",
        "version": 1,
        "domain": "zwave",
        "title": "Z-Wave",
        "data": {
          "network_key": "",
          "usb_path": "/dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_C0F003F0-if00-port0"
        },
        "options": {},
        "pref_disable_new_entities": false,
        "pref_disable_polling": false,
        "source": "user",
        "unique_id": null,
        "disabled_by": null
      },
      {
        "entry_id": "e8ba308ef3ab48a8ab627b59ec63f35c",
        "version": 2,
        "domain": "zha",
        "title": "/dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_C0F003F0-if01-port0",
        "data": {
          "device": {
            "path": "/dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_C0F003F0-if01-port0"
          },
          "radio_type": "ezsp"
        },
        "options": {
          "custom_configuration": {
            "zha_options": {
              "enable_identify_on_join": true
            }
          }
        }
      }
    ]
  }
}```

sreknob, 2021-11-01:
@Hydro-guy - this is how you should reference the alternative baudrate:

```yaml
            {
                "entry_id": "abcdef0123456789abcdef01236456789",
                "version": 3,
                "domain": "zha",
                "title": "/dev/ttyUSB1",
                "data": {
                    "device": {
                        "baudrate": 115200,
                        "path": "/dev/ttyUSB1"
                    },
                    "radio_type": "ezsp"
                },
                "options": {},
                "pref_disable_new_entities": false,
                "pref_disable_polling": false,
                "source": "import",
                "unique_id": null,
                "disabled_by": null
            },
```