walthowd/husbzb-firmware #31: Exception: <type 'exceptions.TypeError'> /dev/ttyUSB1

magic7s, 2021-09-02:
I'm getting a TypeError when trying to scan GoControl HUSBZB-1. Scan fails, the flash seems to complete, but still unable to see anything on zigbee2MQTT. 

OS: Raspbian GNU/Linux Buster (10) on Raspberry Pi 3b+

```
pi@homebridge:~/husbzb-firmware $ python2.7 ./ncp.py scan
Connecting to.. /dev/ttyUSB1 57600 True False
Exception: <type 'exceptions.TypeError'> /dev/ttyUSB1
Connecting to.. /dev/ttyUSB0 57600 True False
^CException: <type 'exceptions.KeyboardInterrupt'> /dev/ttyUSB0
Exception: <type 'exceptions.IndexError'> /dev/ttyAMA0
{"ports": [{"deviceType": "unknown", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}, {"deviceType": "unknown", "pid": "8A2A", "port": "/dev/ttyUSB0", "vid": "10C4"}, {"deviceType": "unknown", "port": "/dev/ttyAMA0"}]}
pi@homebridge:~/husbzb-firmware $ python2.7 ./ncp.py flash -p /dev/ttyUSB1 -f ncp-uart-sw-6.7.8.ebl
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962

Successfully restarted into bootloader mode! Starting upload of NCP image...
Finished!
Rebooting NCP...
pi@homebridge:~/husbzb-firmware $ python2.7 ./ncp.py scan
Connecting to.. /dev/ttyUSB1 57600 True False
Exception: <type 'exceptions.TypeError'> /dev/ttyUSB1
Connecting to.. /dev/ttyUSB0 57600 True False
^CException: <type 'exceptions.KeyboardInterrupt'> /dev/ttyUSB0
Exception: <type 'exceptions.IndexError'> /dev/ttyAMA0
{"ports": [{"deviceType": "unknown", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}, {"deviceType": "unknown", "pid": "8A2A", "port": "/dev/ttyUSB0", "vid": "10C4"}, {"deviceType": "unknown", "port": "/dev/ttyAMA0"}]}
```

magic7s, 2021-09-03:
Doing a bit of poor mans debugging...

At line [209](https://github.com/walthowd/husbzb-firmware/blob/master/ncp.py#L209) the response is `['\x1a', '\xc1', '\x02', '\x0b', '\n', 'R', '~']` which returns `isV8 = True`.

Then at [441](https://github.com/walthowd/husbzb-firmware/blob/master/ncp.py#L441) the response is `['\x12', 'C', '\xa1', '\xa9', '\xfe', '*', '\x15', '\xb5', ',', '\x95', 'L', '"', '\xa2', 'U', '8', '\x95', '\x14', '~']`

So [444](https://github.com/walthowd/husbzb-firmware/blob/master/ncp.py#L444) returns a `<type 'exceptions.TypeError'>`

magic7s, 2021-09-03:
I was able to solve my actual problem of zigbee2mqtt not working with the HUSBZB device. It was actually the default baud speed in zigbee2mqtt. Here are the settings that worked. I still did not get the firmware version to show, but the upgrade was successful as you can see from my original output. So I don't know if an update was needed or not. Feel free to close this issue, or not.  

```yaml
serial:
  port: /dev/ttyUSB1
  adapter: ezsp
advanced:
  baudrate: 57600
```

doron1, 2022-07-29:
> At line [209](https://github.com/walthowd/husbzb-firmware/blob/master/ncp.py#L209) the response is `['\x1a', '\xc1', '\x02', '\x0b', '\n', 'R', '~']` which returns `isV8 = True`.
>
> Then at [...]

I'm seeing the exact same situation, with the exact same response from the stick. Any idea as to how to resolve?

B0ndo2, 2022-12-02:
> I was able to solve my actual problem of zigbee2mqtt not working with the HUSBZB device. It was actually the default baud speed in zigbee2mqtt. Here are the settings that worked. I still did not get [...]

I tried this and it is not working for me. It causes the addon to be stuck showing as started but the frontend displays a blank page
root@8513c2d0f1e7:/tmp/silabs# **./ncp.py scan**
Connecting to.. /dev/ttyUSB1 57600 True False 
{"ports": [{"port": "/dev/ttyUSB1", "vid": "10C4", "pid": "8A2A", "deviceType": "zigbee", "stackVersion": "6.7.8-373"}]}