walthowd/husbzb-firmware #33: Hang after reboot to bootloader with default HUSBZB-1 firmware

henryptung, 2021-09-26:
When using `ncp.py flash` script with an unupdated/original HUSBZB-1 firmware/bootloader, I get a hang at:

```
# ./ncp.py flash -p /dev/ttyUSB1 -f ncp-uart-sw-6.7.8.ebl
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962

```

and it hangs forever in that state until Ctrl-C'd. After much debugging and tweaking of the script, figured out the patch to make it work in my case:

```
    # Open new connection
    ser = serial.Serial(
        port=port,
        baudrate=115200,
        parity=serial.PARITY_NONE,
        stopbits=serial.STOPBITS_ONE
    )
```

My stick needed no flow control at all (either hardware or software), and the `xonxoff=True` flag was causing it to hang when waiting for `1` to be echoed back. Removing that flag allowed the flasher to work.

Putting this here in case this helps other people or this could help improve the flasher script.

MattWestb, 2021-09-26:
Great findings !!

Wot i knowing is all "standard" Silabs bootloaders (EM35X and EFR32 if not cooking one self with other settings) is using no flow control and baudrate 115200.

mloffer, 2021-11-14:
I also ran into this going from 6.6.5 -> 6.7.8. Removing `xonxoff` from the script did the trick.

CTOJoe, 2021-11-26:
Was experiencing the same issue. Removing `xonxoff` from the `# Open new connection` block fixed the issue for me.

dkmcgowan, 2022-01-30:
this happened to me as well, I changed the same line to disable xonxoff just for the one connection where it was not using the CEL variable for xonxoff, seems to have worked fine for me

thechickenmoo, 2022-03-27:
Same. Had to unplug/replug the adapter after my initial hang/fail, restart the docker container, install nano and modify the file. 
I also mistakenly edited the wrong section the first time. (#Init Serial Port Section instead of #Open New Connection)

This did indeed fix the problem.

lefebp, 2022-03-31:
Had the same issue, commenting xonxoff=True also worked for me.

Thank you!

synth3tk, 2022-05-04:
Just chiming in to say this fix also worked for me!

brandon-dacrib, 2022-05-12:
+1 same fix today, can I submit a PR so that the next person doesn't have this problem?

btyeung, 2022-05-22:
+1 also worked for me, linux (kubuntu) with nortek

GroupeFMY, 2022-05-27:
WOW thanks waisted so much time before finding this little info. Thanks works a1

mwpastore, 2022-09-19:
Oh man, I thought I had a bad flash and I've been trying to short those two frustratingly tiny pads for the past hour to get into the bootloader via minicom. I got it to flash after this patch... thank you!

quincarter, 2023-01-08:
i can confirm this works. Just follow what these people have said. :) Thanks a bunch. I was hung up here. Glad i dove into the issues section. 

```shell
root@d74e4d43563d:/tmp/silabs# ./ncp.py flash -p /dev/ttyUSB1 -f ncp-uart-sw-6.7.8.ebl
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962

Successfully restarted into bootloader mode! Starting upload of NCP image...
Finished!
Rebooting NCP...
root@d74e4d43563d:/tmp/silabs# ./ncp.py scan
Connecting to.. /dev/ttyUSB1 57600 True False
{"ports": [{"port": "/dev/ttyUSB1", "vid": "10C4", "pid": "8A2A", "deviceType": "zigbee", "stackVersion": "6.7.8-373"}]}
```

derekgorcz, 2023-01-26:
Also worked for me. Rather than installing nano or vim, you can also use the already installed `sed` to delete the line(for me 322).

1. You can verify line number with this command:
```bash
root@cc611c1f6039:/tmp/silabs# sed -n 322p ncp.py 
        xonxoff=True
```
2. If you see the "xonxoff=True" response, delete it with:
```bash
root@cc611c1f6039:/tmp/silabs# sed -i 322d ncp.py 
```
3. Verify the delete worked:
```bash
root@cc611c1f6039:/tmp/silabs# sed -n 322p ncp.py 
    )
```
4. Run the script as documented.

quietsnow, 2023-04-08:
I have the Home Assistant Blue, so I had to add a number of packages (in working memory, since it runs on Alpine Linux). Here are the series of commands that got this working (and of course, I had to comment out the single line 322 that used to say 'xonxoff=True' along w/ the ',' in the line above - thank you, this was key)...
```
git clone https://github.com/walthowd/husbzb-firmware.git && cd husbzb-firmware
apk update
apk add python3
apk add python3 py3-pip
apk add py3-pyserial
pip3 install xmodem
ln -s /usr/bin/python3 /usr/bin/python
./ncp.py scan
./ncp.py flash -p /dev/ttyUSB1 -f ncp-uart-sw-6.7.8.ebl

```

samuelmcconnell, 2023-10-02:
Another note that ncp.py was hanging when attempting to restart the stick into bootloader mode, and commenting out xonxoff allowed me to flash the firmware successfully.

fireheadman, 2023-10-29:
+1 worked for me also....

xanderphillips, 2023-12-02:
I have had issues with the USB port not apparently providing enough power causing it to not work.  Using a powered hub seems to address the issue quite well.

jeremyforest, 2024-01-22:
+1 worked for me also