walthowd/husbzb-firmware #35: EmberZNet 6.10 / 6.10.x.x Zigbee Coordinator firmware images?

Hedda, 2021-11-16:
Silicon Labs EmberZNet 6.10.3.0 GA should really be mature and stable enough now as only contain bug-fixes since 6.10.0.0 GA.

Silabs EmberZNet Zigbee 6.10.0.0 was released on the 16th June of 2021 and newer EmberZNet 6.10 updates since then:

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.0.0.pdf

EmberZNet 6.10.2.0 was released on the 8th of September of 2021

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.2.0.pdf

EmberZNet 6.10.1.0 was released on the 21st June of 2021

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.1.0.pdf

EmberZNet 6.10.3.0 was released on the 13th October of 2021

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.3.0.pdf

FYI, Elelabs has now released EmberZNet v6.10 firmware for their EFR32 based adapters (based on EFR32MG12):

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/blob/master/data/EFR32MG13/ELE_MG13_zb_ncp_115200_610_211112.gbl

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/commit/31f8e9c09d612734edbe41619240162fe595ecc5

### Added

- 6.10 Zigbee Firmware for ELU013/ELR023 (just keep up with the latest EmberZnet SDK)

Their firmware update tool has also been updated with a few new features:

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/commit/721019cb542b273159372357a30251330e7c8d51

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/commit/e9b3fe1dd73b46fef1629a7a54ca0a257c1388ae

### Changed

- `ele_update` parameters changed. Now the function only updates to latest version of available Zigbee or Thread firmware
- `probe` and `restart` functions can now detect Thread adapters as well

Hedda, 2021-11-25:
> FYI, Elelabs has now released EmberZNet v6.10 firmware for their EFR32 based adapters (based on EFR32MG12):
> 
> https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/blob/master/data/EFR32MG13/ELE_MG13_zb_ncp_115200_610_211112.gbl
> 
> [Elelabs/elelabs-zigbee-ezsp-utility@31f8e9c](https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/commit/31f8e9c09d612734edbe41619240162fe595ecc5)

FYI, Sonoff has now also released Silabs EmberZNet 6.10 (v6.10.3) firmware for ITead Zigbee 3.0 USB Dongle (EFR32MG21 based):

[https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle](https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle)

xsp1989 also announced the good news is that a router firmware "will be released soon" for the same USB dongle here:

[https://github.com/xsp1989/zigbeeFirmware/issues/18](https://github.com/xsp1989/zigbeeFirmware/issues/18)

sreknob, 2021-11-25:
Thanks Hedda for keeping tabs on this. If someone (aka @walthowd @grobasoz) is willing to compile for HUSBZB-1 I’m happy to test!

Hedda, 2021-11-26:
I believe that only EFR32MGxx MCU chips may use EmberZNet 6.8 and later or do I remember that incorrectly?

Anyway, HUSBZB-1 uses the old Silicon Labs EM3581 and Silabs keep removing older chips from each new EmberZNet version.

Zigbee EmberZNet SDK 6.7.10.0 GA way very well be the last version that Silabs EM358x support if they do not release further bug-fix versions (e.g. EmberZNet SDK 6.7.11.0, etc. in the future)

https://cn.silabs.com/documents/public/release-notes/emberznet-release-notes-6.7.10.0.pdf

MattWestb, 2021-11-26:
As Gary have saying Silabs have doping the active support for EM35X devices and also EFR32MG1X and have moving to one SDK that  is not being fixed for the old hardware so its not easy getting it working but not impossible but need many hours / weeks of back porting code that is no grantee that is 110% working in real.

I think EM35X is the last golden release 6.7.10 (perhaps one 11 is coming with some critical bug and security fixes but not knowing).

MG1X is still possible using then its have so much common with the MG1X with large flash / ram that is still active supported but very tricky fitting in the 256k flash.

I have cooking  6.10 multi rail and GPP fixes and its running OK on my Billy in my test network but i dont knowing is its good running on old hardware then comparing with the 6.7.10.X.

6.10 on LIDL/tuya MG21 module is working OK and i was cooking it for 5 mouth ago https://github.com/MattWestb/EFR32-FW/tree/main/MG21LD.
```
EmberZNet version: 6.10.0.0 build 169
```

Hedda, 2021-11-26:
> As Gary have saying Silabs have doping the active support for EM35X devices and also EFR32MG1X

I did not know that Silicon Labs were already dropping support for EFR32MG1X (EFR32MG12/EFR32MG13).

I guess it makes sense since they said they plan on a lower price on EFR32MG21 + manufacturing ramping up for it and MGM210 (EFR32MG21 based MGM210P / MGM210L radio modules), and also EFR32MG23 will be coming soon (should have already been launched if it was not for the chip shortages).

MattWestb, 2021-11-26:
The large (with 7xx K flash) was extended but i dont member then the EOL was being posted but its beyond 6.8.

You dont have the info but  the MG24 support in on the way if reading the closed dev papers that is anti speeding agreement on.

Hedda, 2021-12-08:
FYI, grobasoz has now also started to compile and release EmberZNet 6.10 / 6.10.x.x firmware images for EFR32 / EFR32MGxx:

https://github.com/grobasoz/zigbee-firmware

That is in addition to EmberZNet 6.10 / 6.10.x.x firmware Elelabs and xsp1989 released for EFR32MG12 and EFR32MG21 respectivly:

https://github.com/xsp1989/zigbeeFirmware

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility