walthowd/husbzb-firmware #36: [REQUEST] EmberZNet 6.7.10.0 (6.7.10.x) Zigbee Coordinator firmware images?

Hedda, 2021-11-26:
FYI; Silicon Labs has released Silabs Zigbee EmberZNet SDK 6.7.10.0 GA bugfix release:

https://cn.silabs.com/documents/public/release-notes/emberznet-release-notes-6.7.10.0.pdf

This obviously superseded and repaces 6.7.9.0, 6.7.8.0, etc. and is compatible with all the same chips.

MattWestb, 2021-11-26:
https://github.com/zigpy/zigpy/discussions/595#discussioncomment-277760

Hedda, 2021-12-08:
@walthowd FYI, grobasoz has started compiling 6.7.10.0 / 6.7.10 firmware for older EM358x/EM3581 and ETRX35x/ETRX357 chips:

https://github.com/grobasoz/zigbee-firmware/commits/master

Hedda, 2021-12-21:
@walthowd FYI, grobasoz has updated the EM3581 versions including the ones in the specific release directoty for walthowd:

https://github.com/grobasoz/zigbee-firmware/tree/master/EM3581/%40walthowd

https://github.com/grobasoz/zigbee-firmware/tree/master/EM3581

I have updated the EM3581 versions including walthowd's specific release. PC5 is used for TX Active and PC6 for nTX Active.

_Originally posted by @grobasoz in https://github.com/grobasoz/zigbee-firmware/issues/17#issuecomment-998395507_

sreknob, 2021-12-22:
@Hedda - I tried the 115k2 variants for my HUSBZB-1 with no luck so far using the script. Left a note on the issue in Gary's repo. Anyone here had success with Gary's files yet?

Hedda, 2021-12-23:
> @Hedda - I tried the 115k2 variants for my HUSBZB-1 with no luck so far using the script. Left a note on the issue in Gary's repo. Anyone here had success with Gary's files yet?

No success flashing firmware to HUSBZB-1 or no luck connecting after flashing? You using Zigbee2MQTT or Home Assistant's ZHA?

Note! Should be noted that there will not ever be any 6.8.x.x or later releases for EM35xx/EM35x as that hardware is end-of-life.

PS: I don't have an EM3581 based adapter myself so I only requested 6.7.10.3 firmware update for it since it's still very popular.

sreknob, 2021-12-23:
No luck flashing.
https://github.com/grobasoz/zigbee-firmware/issues/17#issuecomment-999722370
Perhaps @walthowd or someone else with the HUSBZB-1 can give it a try but I think it's an image compiling problem as it was when we first tried Gary’s 6.7.8 firmware before.

sreknob, 2022-01-10:
Good news to report!

Gary recompiled it and I was able to flash 6.7.10 to my HUSBZB-1!

[This is the file I used](https://github.com/grobasoz/zigbee-firmware/blob/master/EM3581/NCP_USW_EM3581_6710.ebl)

```
pi@raspberrypi:~/husbzb-firmware $ ./ncp-115k2.py scan
Connecting to.. /dev/ttyUSB1 115200 True False 
Connecting to.. /dev/ttyUSB0 115200 True False 
{"ports": [{"stackVersion": "6.7.10-423", "deviceType": "zigbee", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}, {"deviceType": "unknown", "pid": "8A2A", "port": "/dev/ttyUSB0", "vid": "10C4"}, {"deviceType": "unknown", "port": "/dev/ttyAMA0"}]}
```

I think the routing settings were at defaults for this one. I know that Gary provided a "walthowd version" for routing configuration specifics with the past versions. Now that it's flashing properly, perhaps he will be kind enough to do that as well.

Follow https://github.com/grobasoz/zigbee-firmware/issues/17 for any further updates he may provide.