walthowd/husbzb-firmware #37: [REQUEST] Support Newer Aqara sensors

stratus-ss, 2021-12-11:
Various Aqara sensors have been updated to version E1.

There is a discussion on the [Home Assistant Forums](https://community.home-assistant.io/t/new-aqara-door-sensors-connection-problem-with-zigbee/351836). Some other controllers such as the ConBeeII have updated firmware to enable this hardware. I'd be willing to chip in a little money to get this support added to the husbzb.

I have some devices and am willing to help test

MattWestb, 2021-12-11:
Too late then its already implanted in bellow but i think  its need more to be added in quirks for this devices.
https://github.com/zigpy/bellows/pull/441

MattWestb, 2021-12-11:
And the quirk ia in place for MCCGQ14LM https://github.com/zigpy/zha-device-handlers/issues/1088.

EZSP coordinators dont need patched firmware then all is don in the host system only deCONZ and TI coordinator must have patched firmware for this devices.

stratus-ss, 2021-12-11:
Thank you for the quick reply!

So for the CECOMINOD016164 HUSBZB-1 it doesn't require updated firmware, I just need to wait for the support to land in HA/upstream projects.

closed 2021-12-11 by stratus-ss