walthowd/husbzb-firmware #38: [REQUEST] em3587 Bitronvideo BV 2010/10

gromgsxr, 2021-12-16:
are you able to add support for em3587 Bitronvideo BV 2010/10.  https://github.com/grobasoz/zigbee-firmware/issues/15 links to the correct firmware and i was able to get it flashed by your code if i edited the pid vid and serial speed to match my stick

originally when i ran the scan i got "vid": "10C4", "pid": "8B34", "deviceType": "unknown"}]} stock speed is 56700 made the edit to ncp.py and i got the current version reported and was able to flash the correct firmware

Hedda, 2021-12-20:
> I managed to upgrade my BV AV2010/10 USB-Stick with the above linked f/w today i used the walthowd docker image and a bit of editing the npc.py

Maybe best if you also submit pull request for review? -> https://github.com/walthowd/husbzb-firmware/blob/master/ncp.py

Hedda, 2021-12-20:
> are you able to add support for em3587 Bitronvideo BV 2010/10. [grobasoz/zigbee-firmware#15](https://github.com/grobasoz/zigbee-firmware/issues/15) links to the correct firmware

FYI, "Bitron Video AV2010/10 USB-stick" is also sold as "SMaBiT AV2010/10 USB-stick", more discussion here ->  https://github.com/zigpy/bellows/issues/348

jogethome, 2021-12-23:
@gromgsxr  , could you please explain, what you've changed in npc.py to get the scan successfully working with the BV 2010/10?
I tried also to add the obvious lines but stuck when reading the version info. 
There's an exception right after 
# EZSP Configuration Frame: getValue ID: 0xAA
# ...
ser.write(b'\x7D\x31\x43\x21\x02\x45\x85\xB2\x7E')
# EZSP Response Frame: getValue ID: 0xAA
# ...
versioninfo = trans(ser.read(16)[1:])[5:]

These are the exceptions I get:
Exception: <class 'NameError'> /dev/ttyUSB0
Exception: <class 'IndexError'> /dev/ttyAMA0

and the type is empty.
Thanks for helping

gromgsxr, 2021-12-23:
i edited the sonoff section to match the correcp pid and vid for the binnatron stick after that it was able to read the firmware version on the stick and also flash the file

# ITEAD/SONOFF EFR32 USB Stick
SONOFF_VID = '10C4'
SONOFF_PID = '8B34'
SONOFF_BAUD = 57600
SONOFF_XON_XOFF = True
SONOFF_RTS_CTS = False

i had to edit the baud rate to 115200 after to confirm the  firmware had updated as the version i used changes the speed

jogethome, 2021-12-23:
Wow, that's a fast answere, thanks :-)
I added a couple of more lines with extra handling of BITRON .. but this is of course faster. I'll try this.
What firmeware did you use, might the NCP_USW_EM3587_6710-LR.ebl from grobasoz the right one (quite new, 3 weeks) ?

gromgsxr, 2021-12-23:
there is a link in my first post ta a thread where i found the file. its NCP_USW_EM3587-LR_678-115k2.ebl

jogethome, 2021-12-23:
ok, it would have been a surprise to me.... the same error.
There must be something else wrong with the Response.
I have got the Firmeware 5.8.0.0 on the stick, that's what is shown in compination of vendor id 4292.0 and produkt id 35636.0 by Openhab  3.2
Do you till have the information what was on yours before the flash? Maybe that would explain the difference.

gromgsxr, 2021-12-23:
stock was 5.8.0.0 and upgraded to 6.7.8-373

gromgsxr, 2021-12-23:
vendor id 4292 is 10C4 in hex and 35636 is 8B34 in hex so same pid and vid. i attempted to make a pull request adding support for the stick

jogethome, 2021-12-23:
ouh yes, than there must be different environmental source of the problem, like version of python, xmodem and/or other imports.
Probably also the fact that it is currently already bound/linked in Openhab (successfully but not really working with ZIGBEE 3.0 items).
I will try to flash the firmware you mention and if I brick the device I will buy another model :-)
If it will work I will give a note here.
Thanks for helping out

gromgsxr, 2021-12-23:
The only thing that may cause problems if you update is the new baud rate as stock it 57600 and the firmware I used was 115200 I don't use openhab so can't comment there. In homeassistant I had to edit the baud to match the new file though

jogethome, 2021-12-23:
That would be possible in Openhab too I guess. But I guess I'll give the other one a try, the one I mentioned

gromgsxr, 2021-12-23:
the version you mentioned flashes to my stick ok i didn't test if it actually works though, and i was able to flash it back the the one i had mentioned

jogethome, 2021-12-25:
the version I was trying was successfully flashed to the stick, even the version came back now with the scan option.
However in Openhab this was not working somehow. Before starting with all the configuraton combinations I tried your first proposal with the 115k ebl file. The flashing worked too and this time the operation inside openhab is still making problems but not as much as the other version before. I guess this is related to openhab. 
During scan of the network for new "things" to bind it seems that the coordinator is somehow falling back to 56k and this on the other side does a knock out of the flashed firmware and it goes offline. However switching manually back to 115k brings it back. The scanning does not work but for now at least a couple of "things" (not all but some) just came back by its own into the openhab "inbox". Strange behaviour.... I'll keep on testing, however this is quite nerving

jogethome, 2021-12-25:
ok, my problem seems to be experienced already by others. 
https://community.openhab.org/t/firmware-upgrade-the-bitronvideo-bv-2010-10-zigbee-usb-dongle/128879/7
Chris Jackson has obviously implemented a fix in the Openhab Zigbee binding ... so I have to find out how to update this

Elix-g, 2021-12-29:
I'm using some Zigbee dongle "BitronVideo AV2010/10", which reports "(Silicon Labs) BV 2010/10" as USB device name and its ID is 10c4:8b34. The stick is sold by German Telekom as accessory for their QIVICON 1 Home Base. Besides, the stick is sold under BitronVideos new company name "SmaBit".

I was finally able to update the stock firmware 5.8.0-134 using update-firmware.sh successfully to 6.7.10. Thanks a lot for your great work!
I've also edited ncp.py slightly in order to detect the stick and read out the stack version properly. After that, I took the risk and updated directly to version 6.7.10 which has been made available by grobasoz:

[https://github.com/grobasoz/zigbee-firmware/blob/master/EM3587/NCP_USW_EM3587_6710-LR.ebl](url)

The dongle so far works smoothly with both bellows and OpenHab. It's indeed just required to define a baud rate of 115200 instead of 57600. In OpenHab, I've got similar issues like jogethome described above but that seems to be an issue in OpenHab indeed, probably because EM3587 supports just a baud rate of 57600.

NilsOF, 2022-01-04:
openHAB 3.2.0 stable should have the baud rate bug fix. This bug existed in snapshots a week or something before it was fixed just before 3.2 RC1.
Fyi: openHAB has a built in coordinator firmware updater in the "openhab-console".

gromgsxr, 2022-01-04:
Don't use openhab I submitted a pull requests to add support for this device on this repo

cz6ace, 2022-01-09:
@gromgsxr , I used the latest snapshot of OpenHab 3.3 and I missed your note to not use that. I did use, it said upgrade started and then nothing happened for fewminutes. After that I restarted OH, changed to 115200 and it started working. **So it works.** I can also confirm that it solved issue with IKEA Tradfri E14 407lm bulb in compare to stock FW of BitronVideo. I tried only the IKEA bulb, the rest of house still running with other stock BV USB stick.

cz6ace, 2022-01-22:
after two weeks I can say the used FW works with my other ZigBee devices OSRAM+ sockets, bulbs, Tuya PIR, IKEA bulbs, Mueller Licht bulbs.

Hedda, 2022-03-14:
FYI, more users confirm EmberZNet 6.7.10 firmware working good on EM3587 here -> https://github.com/grobasoz/zigbee-firmware/issues/15

curlyel, 2024-12-29:
Hi @Elix-g,
may I ask you, *how* exactly you managed to upgrade the firmware. I know, it's a quite old thread, but anyway...

I'm about to upgrade the BV2010/10 using the docker image. I've changed ID settings in the "SONOFF" section as mentioned above and am able to retrieve the current firmware version:
```
root@aa925b818efa:/tmp/silabs# ./ncp.py scan
Connecting to.. /dev/ttyUSB0 57600 True False 
{"ports": [{"port": "/dev/ttyUSB0", "vid": "10C4", "pid": "8B34", "deviceType": "zigbee", "stackVersion": "5.8.0-134"}]}
```
Then, I've downloaded the firmware image into the container:
```
root@aa925b818efa:/tmp/silabs# wget https://github.com/grobasoz/zigbee-firmware/blob/master/EM3587/NCP_USW_EM3587_6710-LR.ebl NCP_USW_EM3587_6710-LR.ebl
--2024-12-29 14:17:26--  https://github.com/grobasoz/zigbee-firmware/blob/master/EM3587/NCP_USW_EM3587_6710-LR.ebl
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘NCP_USW_EM3587_6710-LR.ebl’

NCP_USW_EM3587_6710-LR.ebl          [ <=>                                                   ] 224.98K  --.-KB/s    in 0.02s   

2024-12-29 14:17:27 (10.5 MB/s) - ‘NCP_USW_EM3587_6710-LR.ebl’ saved [230379]
```
There it is:
```
root@aa925b818efa:/tmp/silabs# ls -l NCP*
-rw-r--r-- 1 root root 230379 Dec 29 14:17 NCP_USW_EM3587_6710-LR.ebl
```
Unfortunately, when I'm going to start the upload, the XMODEM throughs errors:
```
root@aa925b818efa:/tmp/silabs# ./ncp.py flash -p /dev/ttyUSB0 -f NCP_USW_EM3587_6710-LR.ebl
Restarting NCP into Bootloader mode...
SONOFF stick
EM3587 Serial Btl v5.8.0.0 b134

Successfully restarted into bootloader mode! Starting upload of NCP image... 
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\n' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'S' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'e' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'i' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b' ' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'u' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'p' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'o' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: NAK received 17 times, aborting.
NCP upload failed. Please reload a correct NCP image to recover.
Rebooting NCP...
```
Luckily, after re-connecting the BV2010/10, the `ncp.py scan` can still retrieve the (stock-) version:
```
root@aa925b818efa:/tmp/silabs# ./ncp.py scan
Connecting to.. /dev/ttyUSB0 57600 True False 
{"ports": [{"port": "/dev/ttyUSB0", "vid": "10C4", "pid": "8B34", "deviceType": "zigbee", "stackVersion": "5.8.0-134"}]}
```
What am I doing wrong?
You said, that you've managed upgrading through the `update-firmware.sh`. But when starting this script, this simply does not find my Zigbee adaptor (again: `ncp.py scan` is still able to _see_ the adaptor):
```
root@aa925b818efa:/tmp/silabs# ./update-firmware.sh 
Did not find compatible zigbee port. Please make sure you passed the correct device through to the docker image
root@aa925b818efa:/tmp/silabs# ./ncp.py scan
Connecting to.. /dev/ttyUSB0 57600 True False 
{"ports": [{"port": "/dev/ttyUSB0", "vid": "10C4", "pid": "8B34", "deviceType": "zigbee", "stackVersion": "5.8.0-134"}]}
root@aa925b818efa:/tmp/silabs#
```

Thanks in advance,
C.

gromgsxr, 2024-12-29:
> Hi @Elix-g, may I ask you, _how_ exactly you managed to upgrade the firmware. I know, it's a quite old thread, but anyway...
>
> I'm about to upgrade the BV2010/10 using the docker image. I've changed [...]

have you edited the ncp.py?  

https://github.com/walthowd/husbzb-firmware/pull/39/commits/c14b92f8d719212bf387e42af3e356e806917ebb

shows what needs editing to make it work with the adaptor i had

curlyel, 2024-12-29:
Thanks for the quick reply!

Well, I just changed the first three lines what you've mentioned in your post.

> SONOFF_VID = '10C4'
> SONOFF_PID = '8B34'
> SONOFF_BAUD = 57600
> SONOFF_XON_XOFF = True
> SONOFF_RTS_CTS = False

So it's getting recognized as "SONOFF" stick. Are the other lines required as well?
It appeared to me, that others had luck with just adopting the VID/PID and BAUD setting...
At least, the bootloader is responding:

```
root@aa925b818efa:/tmp/silabs# ./ncp.py flash -p /dev/ttyUSB0 -f NCP_USW_EM3587_6710-LR.ebl
Restarting NCP into Bootloader mode...
SONOFF stick
EM3587 Serial Btl v5.8.0.0 b134

Successfully restarted into bootloader mode! Starting upload of NCP image... 
```

But for some reason, the XMODEM ack's are bad.

One question though: I needed to pass /`dev/ttyUSB**0**` to the docker container (instead of `/dev/ttyUSB**1**` as in your example). Is the flash operation/XMODEM upload hard wired to USB1 somehow?

C.

Elix-g, 2024-12-29:
> Thanks for the quick reply!
>
> Well, I just changed the first three lines what you've mentioned in your post.
>
>
> So it's getting recognized as "SONOFF" stick. Are the other lines required as well? It [...]

To be honest, I don't remember all the details anymore...did it ages ago. In fact, I didn't use any docker image to upgrade the stick. I also remember, that the hardware isn't capable of 115k baudrate officially but the latest compatible firmware runs at that speed by default. So what you should try is to flash the new firmware with 56k (default of the old hardware and firmware) and switch to 115k after flashing the latest firmware.
Btw: I still use this really old hardware nowadays in Homeassistant and it's still fine even with latest Zigbee devices. The network is stable and even device firmware upgrades are no real challenge.

curlyel, 2024-12-30:
Just checked the ncp.py to understand the flash procedure. There I found:
[image]

... which re-opens a serial port with 115k. It's the same in the original as well as in the modified version of ncp.py.
Is this required for the bootloader or is it needed to set to 57600 baud for the BV2010/10?

gromgsxr, 2024-12-30:
You will need to map the correct usb port to the container. And from memory mine was an old few was 576 originally and updated to 115k after the flash

curlyel, 2024-12-30:
Thanks mates for all the suggestions. But the port device mapping into the container seems not to be the issue, as I'm able to talk to the dongle using ncp.py scan. Meanwhile I've changed the ncp.py to the version @gromgsxr pointed out. 

Using it, the dongle is recognized as BV2010/10 by ncp.py scan but flashing still fails:
```

root@ad31e6b35d91:/tmp/silabs# ./ncp.py scan
Connecting to.. /dev/ttyUSB0 57600 True False 
{"ports": [{"port": "/dev/ttyUSB0", "vid": "10C4", "pid": "8B34", "deviceType": "zigbee", "stackVersion": "5.8.0-134"}]}
root@ad31e6b35d91:/tmp/silabs# ./ncp.py flash -p /dev/ttyUSB0 -f NCP_USW_EM3587_6710-LR.ebl
Restarting NCP into Bootloader mode...
BV2010 stick
EM3587 Serial Btl v5.8.0.0 b134

Successfully restarted into bootloader mode! Starting upload of NCP image... 
ERROR:xmodem.XMODEM:send error: expected ACK; got b'C' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\x18' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'\n' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'S' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'e' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'r' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'i' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'a' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b' ' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'u' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'p' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'l' for block 1
ERROR:xmodem.XMODEM:send error: expected ACK; got b'o' for block 1
ERROR:xmodem.XMODEM:send error: NAK received 17 times, aborting.
NCP upload failed. Please reload a correct NCP image to recover.
Rebooting NCP...
root@ad31e6b35d91:/tmp/silabs#
```

gromgsxr, 2024-12-30:
Why don't you just run it from python why do you need to use docker?

curlyel, 2024-12-30:
Already considered that, but for now, I'm not able to provide a working Python2.7 installation with pySerial and so on...
Therefore the Docker came into play...

curlyel, 2024-12-30:
Unfortunately, the dongle seems to be broken now. LED remains unlit and the ncp.py is no longer able to scan the version.
Have to order a new Zigbee coordinator.
Thanks again for your patience and suggestions.