walthowd/husbzb-firmware #42: ImportError: No module named serial

Stuke00, 2022-03-18:
I have the Nortek HUSBZB-1 USB adapter connected to my machine but I am getting the error "ImportError: No module named serial".  I can see it on /dev/ttyUSB1 but when I run the ncp.py scan script, i get this error.  I am not using docker, running it directly with python on an ubuntu server.  What am I missing here?

baylanger, 2022-03-18:
Seems you need to install the Python serial module.  pip install serial?!

baylanger, 2022-03-18:
@Stuke00 you can also ask google, all you need to do is put this in the search : ImportError: No module named serial

Stuke00, 2022-03-18:
ahh let me try that, I thought it was a script specific error.

Stuke00, 2022-03-18:
ok I gave up on Python and went with a docker setup.  I am still having issues getting this to work.. I verified my device is connected to ttyUSB1, this is what I see on the host system.  

P: /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.1/ttyUSB1/tty/ttyUSB1
N: ttyUSB1
L: 0
S: serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_C1300297-if01-port0
S: serial/by-path/pci-0000:00:14.0-usb-0:2:1.1-port0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.1/ttyUSB1/tty/ttyUSB1
E: DEVNAME=/dev/ttyUSB1
E: MAJOR=188
E: MINOR=1
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=6384794
E: ID_BUS=usb
E: ID_VENDOR_ID=10c4
E: ID_MODEL_ID=8a2a
E: ID_PCI_CLASS_FROM_DATABASE=Serial bus controller
E: ID_PCI_SUBCLASS_FROM_DATABASE=USB controller
E: ID_PCI_INTERFACE_FROM_DATABASE=XHCI
E: ID_VENDOR_FROM_DATABASE=Silicon Labs
E: ID_MODEL_FROM_DATABASE=7 Series/C210 Series Chipset Family USB xHCI Host Controller
E: ID_VENDOR=Silicon_Labs
E: ID_VENDOR_ENC=Silicon\x20Labs
E: ID_MODEL=HubZ_Smart_Home_Controller
E: ID_MODEL_ENC=HubZ\x20Smart\x20Home\x20Controller
E: ID_REVISION=0100
E: ID_SERIAL=Silicon_Labs_HubZ_Smart_Home_Controller_C1300297
E: ID_SERIAL_SHORT=C1300297
E: ID_TYPE=generic
E: ID_USB_INTERFACES=:ff0000:
E: ID_USB_INTERFACE_NUM=01
E: ID_USB_DRIVER=cp210x
E: ID_PATH=pci-0000:00:14.0-usb-0:2:1.1
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_2_1_1
E: ID_MM_CANDIDATE=1
E: DEVLINKS=/dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_C1300297-if01-port0 /dev/serial/by-path/pci-0000:00:14.0-usb-0:2:1.1-port0
E: TAGS=:systemd:

I run docker run --rm --device=/dev/ttyUSB1:/dev/ttyUSB1 -it walthowd/husbzb-firmware bash

then inside the docker container I get no output that the device is found.
root@0cc67379d5c5:/tmp/silabs# ./ncp.py scan
Connecting to.. /dev/ttyUSB1 57600 True False 
{"ports": [{"port": "/dev/ttyUSB1", "vid": "10C4", "pid": "8A2A", "deviceType": "zigbee", "stackVersion": "5.4.1-194"}]}
root@0cc67379d5c5:/tmp/silabs#

baylanger, 2022-03-18:
Isn't this the device?

```
{"ports": [{"port": "/dev/ttyUSB1", "vid": "10C4", "pid": "8A2A", "deviceType": "zigbee", "stackVersion": "5.4.1-194"}]}
```

deviceType = zigbee

Stuke00, 2022-03-18:
I'm not sure, the readme on the main page here shows it should say something after that indicating it found the device.  I pasted the text I am expecting to see below. I do not get the message a device is found, and when I run update_firmware.sh I get the message "Did not find compatible zigbee port. Please make sure you passed the correct device through to the docker image"

root@8b763a33a0d5:/tmp/silabs#./ncp.py scan
{"ports": [{"stackVersion": "5.4.1-194", "deviceType": "zigbee", "pid": "8A2A", "port": "/dev/ttyUSB1", "vid": "10C4"}]}
**Found zigbee port at /dev/ttyUSB1 running 5.4.1-194**

lscorcia, 2022-06-10:
Same issue here. Did you find any solution? (I'm in a VirtualBox ubuntu VM if that makes any difference)

Stuke00, 2022-06-10:
I did not. Ended up getting a new usb adapter. Sent from my iPhone