walthowd/husbzb-firmware #44: [COMPATIBILITY] Can anyone confirm comatibility with ITead’s new “Sonoff Zigbee 3.0 USB Dongle Plus V2” model “ZBDongle-E” based on Silicon Labs EFR32MG21

Hedda, 2022-08-26:
Has anyone confirmed if this Zigbee coordinator firmware updater image from @walthowd is compatible with ITead’s new “Sonoff Zigbee 3.0 USB Dongle Plus V2” model “ZBDongle-E” based on Silicon Labs EFR32MG21

ITead just released a new "**ZBDongle-E**" adapter as an improved version of their [old/previous barebone Silabs EFR32MG21 based Zigbee 3.0 USB adapter from ITead](https://community.home-assistant.io/t/itead-zigbee-3-0-usb-dongle-stick-adapter-based-on-silicon-labs-efr32mg21/271144) and this new variant is to be sold side-by-side as an alternative to [ITead's TI CC2652P based "**Sonoff Zigbee 3.0 USB Dongle Plus**"](https://community.home-assistant.io/t/iteads-sonoff-zigbee-3-0-usb-dongle-plus-based-on-texas-instruments-cc2652p-20dbm-radio-mcu-now-sold-for-14-99/340705) (which is now renamed to "**ZBDongle-P**"), meaning they will continuously sell both variants. 

[https://itead.cc/product/zigbee-3-0-usb-dongle/](https://itead.cc/product/zigbee-3-0-usb-dongle/)

[https://sonoff.tech/product-review/sonoff-zigbee-3-0-usb-dongle-plus-tutorials/](https://sonoff.tech/product-review/sonoff-zigbee-3-0-usb-dongle-plus-tutorials/)

[https://sonoff.tech/wp-content/uploads/2022/08/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing-.pdf](https://sonoff.tech/wp-content/uploads/2022/08/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing-.pdf)

[image]

This new “Sonoff Zigbee 3.0 USB Dongle Plus V2” adpater can be a little confusing to buyers as it looks the same as Sonoff Zigbee 3.0 USB Dongle Plus V2” and both this "**ZBDongle-E**" variant and the "**ZBDongle-P**" variant is market as being able to act either a Zigbee Coordinator (by default) or Zigbee Router device (if flashed with such firmware instead), and if flash [OpenThread/Spinel Radio Co-Processor (RCP)](https://openthread.io/platforms/co-processor) firmware also [Thread network protocol](https://en.wikipedia.org/wiki/Thread_(network_protocol)) for the upcoming [Matter/CHIP standard (Project Connected Home over IP)](https://en.wikipedia.org/wiki/Matter_(standard)). As Silicon Lab and Texas Instruments adapters offer different compatibility ITead will now sell both of these so users will now have more options as different DIY home automation software applications offer might not be fully compatible with one or the other (Home Assistant's ZHA integration is however fully compatible with both so based in just the specification they should in theory offer similar performance on paper).

### Comparing "ZBDongle-P" vs. "ZBDongle-E" vs. barebone EFR32MG21 dongle

https://community-assets.home-assistant.io/original/4X/2/9/d/29dc429141353119ad34f33516ed61f5fc2b3af3.png  verses  
[image]
 verses  
[image]

|Feature/Model |ZBDongle-P|ZBDongle-E|9888010100045|
| --- | --- | --- |--- |
|Radio SoC/MCU chip|Texas Instruments CC2652P|Silicon Labs EFR32MG21|Silicon Labs EFR32MG21|
|Zigbee Stack (Serial Interface Protocol API/CLI)|Z-Stack v3 (ZNP 3)|EmberZNet (EZSP v8)|EmberZNet (EZSP v8)|
|Optional Zigbee Router firmware|Yes (9dBm firmware available from Koenkk)|Yes (20dBm firmware available from ITead)|Yes (20dBm firmware available from ITead)|
|USB to UART/Serial Converter Chip|CP2102 or CP2102N|CH9102F|CH340|
|USB EEPROM Product Description ID| SONOFF Zigbee 3.0 USB Dongle Plus| SONOFF Zigbee 3.0 USB Dongle Plus V2|None (no USB chip EEPROM)
|Home Assistant USB Auto Disovery|Yes|Not yet|Not possible|
|Flow Control|None by default (Hardware flow control optional with alternative firmware and flipped dip-switch）|Software flow control|Software flow control|
|RF Transmit Output Power |9dBm (firmware hardcoded), Max: 20dBm|20dBm (default)|20dBm (default)|
|Antenna|External (rotatable and tiltable)|External (rotatable and tiltable)|Onboard circuit board antenna|
|Enclosure/case|Aluminum all-metal shell casing|Aluminum all-metal shell casing|None|
|Length|63mm|52mm|?|
|Packaging|Retail-box with manual|Retail-box with manual|Anti-static bag only|

Both their Silicon Labs and Texas Instruments adapters are compatible out-of-the-box with Home Assistant's [ZHA integration](https://www.home-assistant.io/integrations/zha) which supports Silabs EZSP (EmberZNet Serial Protocol) as well as TI Z-Stack Serial Interface and even more manufacturers Zigbee adapters, so different benefits are that the same Silicon Labs EFR32MG21 based adapter is also used by [Home Assistant SkyConnect USB Stick](https://community.home-assistant.io/t/home-assistant-skyconnect-usb-stick-announced-will-be-compatible-with-both-zigbee-and-thread-including-matter-chip-over-thread/433594) (as well as [ITead's Sonoff ZBBridge](https://community.home-assistant.io/t/sonoff-zbbridge-sonoff-zigbee-bridge-from-itead/187346) and [ZB-GW03 eWeLink Ethernet Zigbee Gateway](https://community.home-assistant.io/t/zb-gw03-ewelink-ethernet-zigbee-gateway-now-hacked-with-tasmota-and-esphome-so-can-be-used-via-mqtt-or-as-a-remote-zigbee-lan-adapter-with-home-assistants-zha-integration-or-zigbee2mqtt/341223) which with Zigbee Coordinator NCP firmware they are compatible with [OpenHAB's ZigBee Binding](https://www.openhab.org/addons/bindings/zigbee/), [Zigbee Plugin for Domoticz](https://github.com/zigbeefordomoticz/Domoticz-Zigbee), and [Zigbee Plugin for Jeedom](https://github.com/zigpy/zigpy/discussions/725), while the Texas Instruments CC2652P based adapter is also fully compatible with [Zigbee2MQTT (a.k.a. Z2M)](https://www.zigbee2mqtt.io/) and [IoBroker](https://github.com/ioBroker/ioBroker.zigbee) while their [zigbee-herdsman library currently only has experimental support for Silicon Labs based adapters](https://github.com/Koenkk/zigbee-herdsman/issues/319).

# Zigbee 3.0 USB Dongle Plus - ZBDongle-E variant

[https://itead.cc/product/zigbee-3-0-usb-dongle/](https://itead.cc/product/zigbee-3-0-usb-dongle/)

It uses the same [EFR32MG21A020F768IM32 SoC](https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-socs/device.efr32mg21a020f768im32?tab=specs) used on the first ITead Zigbee 3.0 USB Dongle based on EFR32 Mighty Gecko Series 2, as well as on CoolKit Technologies "SM-011 V1.0" module used in the ITead Sonoff ZBBridge Zigbee Bridge and in Tuya ZS3L and Tuya ZSLC5 SoC modules. It includes a 80 MHz ARM Cortex-M33 core, 768 Flash Storage, 64 RAM, in a QFN32 package, which radio has + 20 dBm maximum power output and receive sensitivity of -104 (250 kbps O-QPSK DSSS) dBm. 

[image]

[image]

darkxst, 2022-08-28:
I have tested. It doesn't recognise the Product ID of (55D4) of the new ZBDongle-E, however after manually patching that Id it detects correctly the device and successfully flashes the image, at least with the [iTead provided firmware images](https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E).

However once flashed with the iTead router firmware, neither this tool nor the Elelabs tool, can get the device back into bootloader mode. So far that only seems possible with the physical boot button.

Hedda, 2022-08-29:
Yeah, this new "ZBDongle-E" adapter apprently uses a WCH CH9102F (CH9102/CH340 device drivers) which has VID (Vendor ID) "1A86 "and PID (Product ID) 55D4 as well as new USB product description identifier "SONOFF Zigbee 3.0 USB Dongle Plus V2":

https://github.com/home-assistant/core/pull/77192

https://community.home-assistant.io/t/itead-s-sonoff-zigbee-3-0-usb-dongle-plus-v2-model-zbdongle-e-based-on-silicon-labs-efr32mg21-20dbm-radio-mcu-now-sold-for-19-99/442695

https://community.home-assistant.io/t/community-help-wanted-to-whitelist-all-compatible-zigbee-and-z-wave-usb-adapters-for-automatic-discovery-in-home-assistant-os/344412