walthowd/husbzb-firmware #47: Problem running the python script.

ovcrash, 2022-11-15:
Hi,

The python script, doesn't seem to detect the device.

└──╼ $./ncp.py scan
Connecting to.. /dev/ttyUSB1 57600 True False 
Exception: <class 'serial.serialutil.SerialException'> /dev/ttyUSB1 
Connecting to.. /dev/ttyUSB0 57600 True False 
Exception: <class 'serial.serialutil.SerialException'> /dev/ttyUSB0 
{"ports": [{"port": "/dev/ttyUSB1", "vid": "10C4", "pid": "8A2A", "deviceType": "unknown"}, {"port": "/dev/ttyUSB0", "vid": "10C4", "pid": "8A2A", "deviceType": "unknown"}]}

I'm pretty sure i have everything setup correctly. Anyone else got this error?

DustStormPettigrew, 2023-01-17:
I was able to complete my flash without getting the "Found zigbee port" message. I just continued on with using /dev/ttyUSB1 as my target (assuming no other USB devices are installed except mouse and keyboard, this looks like it's pretty consistently the right device). I also had to follow the steps in #33 but after that, it seemed to work.

mimimi0815, 2023-03-11:
you have to install the stuff.
pip3 install pyserial
and
pip3 install xmodem

(maybe with "sudo" prefix
after that start script agian

stamandr, 2023-08-01:
I am getting the error "Permission denied" running the command below. I have installed pyserial and  xmodem. Any suggestions ?

root@NUC:~# ./ncp.py scan
bash: ./ncp.py: Permission denied