walthowd/husbzb-firmware #49: Nortec combo stick not found.

IanAber, 2022-12-15:
I have a NortecHUSBZB-1 stick. It is recognised as ttyUSB0 and ttyUSB1 in my Linux machine but the ncp.py scan command fails to find it.

Exception: <class 'IndexError'> /dev/ttyS0 
Connecting to.. /dev/ttyUSB1 57600 True False 
Connecting to.. /dev/ttyUSB0 57600 True False 
No ZigBee Ack. /dev/ttyUSB0 
{"ports": [{"port": "/dev/ttyS0", "deviceType": "unknown"}, {"port": "/dev/ttyUSB1", "vid": "10C4", "pid": "8A2A", "deviceType": "zigbee", "stackVersion": "5.4.1-194"}, {"port": "/dev/ttyUSB0", "vid": "10C4", "pid": "8A2A", "deviceType": "unknown"}]}

DustStormPettigrew, 2023-01-17:
Looks like found it at /dev/ttyUSB1 based on the ports JSON at the end. Looks like it's just missing the "Found zigbee port" message.

DustStormPettigrew, 2023-01-17:
I just specified ttyUSB1 on the next step, commented the line mentioned in issue #33