walthowd/husbzb-firmware #51: Starts bootloader but doesn't recognize success?

ndrwstn, 2023-03-03:
I'm having the same issue as #50 (but I'm posting a new issue because the Cyrillic topic probably isn't helpful to most users, no offense intended).

I'm trying to flash ncp-uart-6.7.8-115k.ebl on  nortek husbzb-1 with 5.4.1-194 (brand new device) using the docker image. After 'CEL stick' I get 'EM3581 Serial Bootloader v.5.4.1.0 b962' followed by a newline and then the script hangs (and I have to kill it). Good news is no damage done to the device, but other than than nothing seems to happen.

I initiatized my device in minicom and it reacts as expected (both from the README and how the script works), and a string of 'C's were received. However, for some reason we don't get to L364 ('Successfully restarted ...') in ncp.py. Looking for suggestions.

https://github.com/walthowd/husbzb-firmware/blob/d6a512ccb2d770d4fdabdc27805ddd68fde96988/ncp.py#L343-L364

mimimi0815, 2023-03-11:
I have the same issue. I assume it has also something todo with the "Scan", because when i do a scan first, i have also to break it.
First ttyUSB1 will be normally detected. (with zigbee), but on ttyUSB0 it hangs and i have to cancel the scirpt.

CMD: ./ncp.py scan
Connecting to.. /dev/ttyUSB1 57600 True False 
Connecting to.. /dev/ttyUSB0 57600 True False 
^CException: <class 'KeyboardInterrupt'> /dev/ttyUSB0 
{"ports": [{"port": "/dev/ttyUSB1", "vid": "10C4", "pid": "8A2A", "deviceType": "zigbee", "stackVersion": "5.4.1-194"}, {"port": "/dev/ttyUSB0", "vid": "10C4", "pid": "8A2A", "deviceType": "unknown"}]}

ndrwstn, 2023-03-11:
Are you trying to scan for the zwave tty or am I misunderstanding? I do not believe this script is supposed to be able to upgrade the zwave firmware--only zigbee.

draftpunk, 2023-03-14:
Referring to this issue resolved my issue
https://github.com/walthowd/husbzb-firmware/issues/33

ndrwstn, 2023-03-17:
The fix in #33 worked for me as well, and as I think there's a PR waiting there, I don't think there needs to be multiple open issues on the same problem. I'm going to close this, and I think #50 should also try this fix and close if it works.