walthowd/husbzb-firmware #54: /usr/bin/env: 'python': No suck file or directory

Tical84, 2023-07-26:
After following all of the steps down to running ./ncp.py scan I get this error message.  I'm trying as root, installed all prerequisite dependencies from what I can tell.  I also found an alternate ncp.py file on another site and gave that a try and got "permission denied" so I went back to the included one.

I'm new on Linux, but can work my way through most things.

stamandr, 2023-08-01:
Same issue for me as well...

root@Deb11:~# ./ncp.py scan
bash: ./ncp.py: Permission denied

MattWestb, 2023-08-01:
Use sudo then the user dont have access to the comports like the root is having or adding the rights tithe user.

MattWestb, 2023-08-02:
For some reason you have not exec rights on the file `./ncp.py` perhaps no execute flag sett on it ?

shilllabs, 2023-11-25:
Had the same issue.  I needed to add a sym link, 

sudo ln -s /usr/bin/python3 /usr/bin/python